andresarmento/modbus-arduino #6: AT command mode not working

heroseller17, 2016-02-08:
hello
i program the esp8266 at command mode lamp.ino on mega2560 and connect it with OPC but its always get Bad data and not responding..:(
what is the problem here?
(i use esp-01 black versian and i dont flash it.)

Alexbox364, 2016-02-21:
Hello, 
I have the same issue than you. I haven't found the solution yet to have it work. I use the same ESP8266 version than yours. I will try another version. and see what happens...

Alexbox364, 2016-02-21:
By the way, many thanks to the author for it great work !!
I tested the ESP8266 on nodeMCU & the ethernet versions & they work like a charm!

heroseller17, 2016-02-21:
hello
you say that using the esp8266 mcu?and program it?

Alexbox364, 2016-02-21:
Hello. 
Among all the versions you made for Modbus TCP/IP, I tried those :
- W5100 arduino Ethernet shield + arduino uno was ok.
- ENC28J60 shield on arduino nano was ok too
- ESP8266AT with arduino nano : I could ping my Modbus slave , establish a TCP connection but never could read any coil or holding register, I always had an data error from my Modbus master (qMod on Windows). Seems to be the same issue as @heroseller17 - I used the same esp8266-01 board. I haven't tried with another one. I haven't tried to update the AT firmware of the ESP8266 neither.
- the last I tried with success was the ESP8266 version. I used a nodeMCU 0.9 board to make it easier to program the ESP8266 chip on it.
Best regards.
Alexandre.

heroseller17, 2016-02-22:
hello 
yes i jsut can say thank you to Mr.samento for this great library....i tried AT-command mode on arduino mega to such as@Alexbox364 too!but cant find that..:(

Alexbox364, 2016-02-28:
Hi, 
I finally got the library to work with the ESP8266 with AT command firmware.
Here is what I did : 
- update the ESP8266 with the latest firmware - Follow this tutorial : http://www.allaboutcircuits.com/projects/update-the-firmware-in-your-esp8266-wi-fi-module/
But I used the latest firmware up to date on the 25th february 2015 that you can find here : http://bbs.espressif.com/viewtopic.php?f=46&t=1702
- Once my ESP8266 was up to date, it still didn't work for me. Apparently because my ESP8266 was set to station mode & Access point. I then slightly modified the library "ModbusIP_ESP8266AT.cpp" on line 15 an changed "wifi.setOprToStationSoftAP()" to "wifi.setOprToStation()". 
- I also changed the file "ModbusIP_ESP8266AT.h" on line 14 and set "#define MODBUSIP_TIMEOUT  10" to "#define MODBUSIP_TIMEOUT  100"

I took me a while to have it work but now I have thanks to Andreas a nice Wifi TCP modbus slave!
My slave is just a bit slow to receive data (aprox. 2 secs) compared to the Ethenet versions (almost instantaneous).

Alexbox364, 2016-02-28:
By the way my hardware is Arduino nano rev 3 + ESP-7 (with antenna) + LM1117MP-3.3 to stepdown voltage from 5V to 3.3V to power up the ESP8266. I haven't made anything special to step-up voltage on Serial interface etween arduino and ESP8266.

heroseller17, 2016-04-17:
@Alexbox364  hello too!
have you ever test this code on project?the esp is unstable and reset many times!!why?????i cant use it in my projects.i have esp8266 esp-07 with antena and lm1117 3.3 and relay...but it resets and when i turn the relay on i get bad data in opc...could you please help me?