andresarmento/modbus-arduino #7: Help with multiple coils/inputs

marcusjelsch, 2016-02-21:
Hi!

1. All thumbs up for this great work.

Now my problem, how to modify the code for use with multiple coils.

`/*

I2C Adressen:

0x20 - Relais
0x21 - Eingänge (5 Taster und 3 Extern)
0x22 - LCD 4x20 Zeichen

*/

#include <Wire.h>
#include <PCF8574.h>
#include <LiquidCrystal_I2C.h>
#include <EtherCard.h>
#include <Modbus.h>
#include <ModbusIP_ENC28J60.h>

LiquidCrystal_I2C lcd(0x22, 4, 5, 6, 0, 1, 2, 3, 7, POSITIVE);  // Set the LCD I2C address

ModbusIP mb;
//PCF8574 Objekt 
PCF8574 inputs;    // PCF8574 mit Adresse 0x21
PCF8574 outputs;    // PCF8574 mit Adresse 0x20

void setup() 
  {
  // The media access control (ethernet hardware) address for the shield
  byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
  // The IP address for the shield
  byte ip[] = { 192, 168, 0, 44 };
  //Config Modbus IP
  mb.config(mac, ip);
  
  for (int i=100;i<8;i++)
    {
    mb.addIsts(i);
    }
  
  for (int i=200;i<8;i++)
    {
    mb.addCoil(i);
    }
    
  io_setup(); // Eingänge und Ausgänge initialisieren

  lcd.begin(20,4);  
  }

void loop() 
{
  // poll messages
  mb.task();
  
   lcd.home (); 
   lcd.print ("Adr.: 192.168.0.44:502");
   lcd.print (" ");
         
  // link the Arduino pins to the Modbus array
  io_poll();
} 

void io_setup() 
  {  
  inputs.begin(0x21);
  for (int i=0;i<8;i++)
    {
    inputs.pinMode(i, INPUT_PULLUP);
    }
  
  outputs.begin(0x20);
  for (int i=0;i<8;i++)
    {
    outputs.pinMode(i, OUTPUT);
    }
  }
  
/**
 *  Link between the Arduino pins and the Modbus array
 */
void io_poll() {
  
  //Digitale Eingange abfragen
  for (int i=100;i<8;i++)
    {
    mb.Ists(i, inputs.digitalRead( i-100 ) );
    }
  
 
  //Digitale Ausgänge
  //Adresse 16-31
  outputs.digitalWrite( 0, mb.Coil(202));
  outputs.digitalWrite( 1, mb.Coil(201));
  outputs.digitalWrite( 2, mb.Coil(200));
  outputs.digitalWrite( 3, mb.Coil(203));
  outputs.digitalWrite( 4, mb.Coil(207));
  outputs.digitalWrite( 5, mb.Coil(206));
  outputs.digitalWrite( 6, mb.Coil(205));
  outputs.digitalWrite( 7, mb.Coil(204));  
  
}`

I can only read the input at adress 100.

What am i doing wrong?

Thank u for help.

marcusjelsch, 2016-02-21:
Done...

I made mistakes in the for-loops!

But now i have the problem, that at adress 107 and at adress 207 i can see/do no changes???!!!

Here my actual code:

`/*

I2C Adressen:

0x20 - Relais
0x21 - Eingänge (5 Taster und 3 Extern)
0x22 - LCD 4x20 Zeichen

*/

#include <Wire.h>
#include <PCF8574.h>
#include <LiquidCrystal_I2C.h>
#include <EtherCard.h>
#include <Modbus.h>
#include <ModbusIP_ENC28J60.h>

LiquidCrystal_I2C lcd(0x22, 4, 5, 6, 0, 1, 2, 3, 7, POSITIVE);  // Set the LCD I2C address

ModbusIP mb;
//PCF8574 Objekt 
PCF8574 inputs;    // PCF8574 mit Adresse 0x21
PCF8574 outputs;    // PCF8574 mit Adresse 0x20

void setup() 
  {
  // The media access control (ethernet hardware) address for the shield
  byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
  // The IP address for the shield
  byte ip[] = { 192, 168, 0, 44 };
  //Config Modbus IP
  mb.config(mac, ip);
  
  for (int i=100;i<108;i++)
    {
    mb.addIsts(i);
    }
  
  for (int i=200;i<208;i++)
    {
    mb.addCoil(i);
    }
    
  io_setup(); // Eingänge und Ausgänge initialisieren

  lcd.begin(20,4);  
  }

void loop() 
{
  // poll messages
  mb.task();
  
   lcd.home (); 
   lcd.print ("IP-Adr:192.168.0.44");
   
         
  // link the Arduino pins to the Modbus array
  io_poll();
} 

void io_setup() 
  {  
  inputs.begin(0x21);
  for (int i=0;i<8;i++)
    {
    inputs.pinMode(i, INPUT_PULLUP);
    }
  
  outputs.begin(0x20);
  for (int i=0;i<8;i++)
    {
    outputs.pinMode(i, OUTPUT);
    }
  }
  
/**
 *  Link between the Arduino pins and the Modbus array
 */
void io_poll() {
  
  //Digitale Eingange abfragen
  for (int i=100 ;i<108;i++)
    {
    mb.Ists(i, inputs.digitalRead( i-100 ) );
    }
  
 
  //Digitale Ausgänge
  //Adresse 16-31
  outputs.digitalWrite( 0, mb.Coil(202));
  outputs.digitalWrite( 1, mb.Coil(201));
  outputs.digitalWrite( 2, mb.Coil(200));
  outputs.digitalWrite( 3, mb.Coil(203));
  outputs.digitalWrite( 4, mb.Coil(207));
  outputs.digitalWrite( 5, mb.Coil(206));
  outputs.digitalWrite( 6, mb.Coil(205));
  outputs.digitalWrite( 7, mb.Coil(204));  
  
}`

jsumm, 2016-09-16:
I imagine you've already figured out a fix for this, but what I've noticed is that for coils and discrete inputs there's a problem with every 8th coil/input when you access them using the multiple read/write MODBUS commands.  If you have the bandwidth you can change from using read/write multiple coils to read/write single coils and loop over all 8 coils or you can read/write 7 or less coils at a time in multiple read/write calls.  

I imagine the library has a bit shift error of some sort, because it happens on every 8th coil (if you were trying to work with 16 you'd have issues with both the 8th and 16th).

DeTommer, 2017-09-05:
It's actually offsetted with 8.
I'm using the following code (together with the PCA9555 shield):
`
/*
  Modbus-Arduino Example - Switch (Modbus IP)
  Copyright by André Sarmento Barbosa
  http://github.com/andresarmento/modbus-arduino
*/
  //Avoid pins 4,10,11,12,13 when using ethernet shield
 
#include <SPI.h>
#include <Ethernet.h>
#include <Modbus.h>
#include <ModbusIP.h>
#include <HardwareSerial.h>
#include <clsPCA9555.h>
#include <Wire.h>

//Modbus Registers Offsets (0-9999)
//const unsigned ISTS0 = 0;

//Modbus Register
const unsigned int IREG0 = 0;

unsigned ioval=0;
unsigned io=0;
//PCA address
PCA9555 ioport(0x20);
#define _IO_CNT 16

//ModbusIP object
ModbusIP mb;
void setup() {
    // The media access control (ethernet hardware) address for the shield
    byte mac[] = { 0x00, 0x06, 0x59, 0x00, 0x1B, 0xE4 };  
    // The IP address for the shield
    byte ip[] = { 172, 16, 47, 10 };   
    //Config Modbus IP 
    mb.config(mac, ip);
	
	for(byte i=0;i<_IO_CNT;i++)
	{
		ioport.pinMode(i, INPUT); //Set I2C GPIOs pinMode input
		mb.addIsts(i);				//add input status register according to ammount of inputs
	}
		
	mb.addIreg(IREG0);
	
  // serial setup
  Serial.begin(9600);
  Serial.println("Serial interface started");
}

void loop() {
   //Call once inside loop() - all magic here
   mb.task();
      
   for(byte i=0;i<_IO_CNT;i++)
   {
	   io = ioport.digitalRead(i);		//read IO value
	   mb.Ists((i),io);					//puts IO value in Input Status
	   ioval+=(io<<i);					//Shifts IO value in 16 bit Input Register
   }

	mb.Ireg(IREG0,ioval);   
   ioval=0;
}
`
What I see is that when I read register 0..7, I read the correct data at 0..6 & 15... Do 7 jumped to 15, the actual 15 has jumped to 23.
[image]

I know this project is presumed dead, but it would still be nice to find a solution or to help others :-).
If I find anything, I will post it here

DeTommer, 2017-09-05:
Fixed by using this file
[https://github.com/seydamir/modbus-arduino/blob/9e124a2ef7e6e06f5ad360691205ef0d04df92a0/libraries/Modbus/Modbus.cpp](url)

pawel3410, 2018-07-06:
**DeTommer **
Could You send again this fixed file?