andresarmento/modbus-arduino #8: Arduino Due

josb86, 2016-03-10:
Hi, Is this library compatible with the Arduino Due? I am traying run a Basic lamp example and I get a error with the hardware serial format, can you help me?

NaKroTeK, 2016-03-24:
Hi josb86, 
If you did not resolve your problem, I had the same problem with this card : Arrow SmartEverything IoT SoM.

I manage to get it work, by tweaking ModbusSerial.cpp
By default it uses this function :
`bool ModbusSerial::config(HardwareSerial* port, long baud, u_int format, int txPin)
`
I tried with : remplace  `_VARIANT_AMEL_SMARTEVERYTHING_` with your card
```
#ifdef _VARIANT_AMEL_SMARTEVERYTHING_
bool ModbusSerial::config(Serial_* port, long baud, u_int format, int txPin) {
....
}
```

To be sure, force the lib to use this function : 
`bool ModbusSerial::config(Serial_* port, long baud, u_int format, int txPin)`
instead of :
`bool ModbusSerial::config(HardwareSerial* port, long baud, u_int format, int txPin)`

josb86, 2016-08-31:
I can´t get that this library run on the DUE please, I tried to do that say NaKroTeK but nothing succeed.

NaKroTeK, 2016-09-01:
I apologise if it was not clear.
Look for something like _VARIANT_AMEL_SMARTEVERYTHING_
But for your arduino due.

josb86, 2016-09-01:
Hi NaKroTeK, I understand your reply and I triyed put in the file "modbusserial.cpp":

```
#ifdef _SAM3XA_
bool ModbusSerial::config(Serial_* port, long baud, u_int format, int txPin) {
    this->_port = port;
    this->_txPin = txPin;
    (*port).begin(baud, format);
    while (!(*port));

    if (txPin >= 0) {
        pinMode(txPin, OUTPUT);
        digitalWrite(txPin, LOW);
    }

    if (baud > 19200) {
        _t15 = 750;
        _t35 = 1750;
    } else {
        _t15 = 15000000/baud; // 1T * 1.5 = T1.5
        _t35 = 35000000/baud; // 1T * 3.5 = T3.5
    }

    return true;
}
#endif
```

and 

```
#ifndef __ARM__
bool ModbusSerial::config(Serial_* port, long baud, u_int format, int txPin) {
    this->_port = port;
    this->_txPin = txPin;
    (*port).begin(baud, format);
    while (!(*port));

    if (txPin >= 0) {
        pinMode(txPin, OUTPUT);
        digitalWrite(txPin, LOW);
    }

    if (baud > 19200) {
        _t15 = 750;
        _t35 = 1750;
    } else {
        _t15 = 15000000/baud; // 1T * 1.5 = T1.5
        _t35 = 35000000/baud; // 1T * 3.5 = T3.5
    }

    return true;
}

#endif
```

At the same in the file "modbusserial.h":

```
#ifdef _SAM3XA_
        bool config(Serial_* port, long baud, u_int format, int txPin=-1);
        #endif
```

and

```
#ifdef __ARM__
        bool config(Serial_* port, long baud, u_int format, int txPin=-1);
        #endif
```

And nothing happen. I got tehe same error:

```
C:\Users\jose\Documents\Arduino\libraries\ModbusSerial\ModbusSerial.cpp:23:32: error: invalid conversion from 'u_int {aka unsigned int}' to 'UARTClass::UARTModes' [-fpermissive]

     (Serial).begin(baud, format);
```

```
C:\Users\jose\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.9\cores\arduino/UARTClass.h:48:10: error:   initializing argument 2 of 'void UARTClass::begin(uint32_t, UARTClass::UARTModes)' [-fpermissive]

     void begin(const uint32_t dwBaudRate, const UARTModes config);
```

idexon, 2018-02-10:
bool ModbusSerial::config(HardwareSerial* port, long baud, u_int format, int txPin) {
    this->_port = port;
    this->_txPin = txPin;
    //(*port).begin(baud, format);
	(*port).begin(baud);

asdelx1, 2019-06-01:
exactly, just delete "format" and it works