andresarmento/modbus-arduino #11: ESP8266 Arduino

cancodr, 2016-04-01:
I have started working on an ESP8266 with native Arduino (ie: NOT AT mode) version of this (at least for IP).  Is this project still alive?  I'd be happy to contribute, but I need a little help / guidance on a few things as I'm new to GitHub.

skefer, 2016-04-09:
I think the project is still going on,  git is very easy to use, you can download tortoisegit on windows and begin your contribution.

Presjar, 2016-04-13:
Hi cancodr. I'm very interested is using a esp8266 to gather modbus data via IP. All the best with your coding. I'd love to try it out!

cancodr, 2016-04-21:
I have a preliminary version (.cpp and .h files).  I have attached it here.  I'm really sorry, I haven't had the time to learn Git, so I'm not sure the proper methodology go contribute.  If there is a good tutorial or something, I'd love to learn.

My code uses the ModbusIP library as a basis, but is modified to run similar to the ESP8266WebServer in terms of how it handles the states.  Documentation is thin, and I haven't had any more time to test or do anything with it as I realized I actually need a TCP/RTU gateway rather than a TCP server....so I'm working on that now.

[ModbusIP_ESP8266_Arduino.zip](https://github.com/andresarmento/modbus-arduino/files/230164/ModbusIP_ESP8266_Arduino.zip)

Happy to get some feedback on this.

Presjar, 2016-04-22:
Hi! I'm pretty new to this stuff as well. Especially modbus. As the documentation for a lot of the libraries as not very complete it is hard to know what exactly to do. I've downloaded your files and will have a look but i'm way behind where you are :p 

I'd be keen to see what code you are using for your .ino arduino stuff as that could help me get what the functions you have written can do for me.

Thanks, Looking forward to working with you in what capacity I can!

cancodr, 2016-04-22:
I will try to put together a simple sketch and attach here.

cancodr, 2016-04-22:
I have attached another ZIP file.  Note, I changed the library slightly.  Also, I noticed that my library is using the serial port for debug output, so you may want to clean it up (ie: line 42, 52, etc).

There is an ino file that shows how to setup the Modbus IP object, and accept connections.  You need to ensure that you run "handleClient()" in your main loop.

I also included a JPG file showing successful interaction using modpoll (free command line software).

You will of course need to input your own SSID and PASSWORD settings to get it on your network.

I'd love to hear how you get along with it.

[WifiModbusIP.zip](https://github.com/andresarmento/modbus-arduino/files/232410/WifiModbusIP.zip)

Presjar, 2016-04-24:
That work very well! I got it to work with the program Modbus Master.

The next thing I need is the ESP8266 to collect data from Holding registers from another modbus device on the network. example. ESP8266 is IP 192.168.0.31 and I want data from IP 192.168.0.32:502 Holding Register 30700. Can this ESP be a client rather than a host? The attached image is an example of what I want to achieve.

[image]