andresarmento/modbus-arduino #30: How Can I add Ethernet2.h library?

huytranthanh1508, 2017-12-14:
I am using W5500 Chip for ethernet communication. However, I have to use ethernet2.h for all examples. 
When I declare ethernet2.h for Lamp or Switch Examples. It shows error with "Error compiling for board Teensy 3.6.". I use teensy 3.6 
Thank you

AdrianTschannen, 2018-01-07:
I am trying also to use Modbus in combination with ethernet2. Probably this could help:
If you have a look at modbus.h code, you will realize that there the library Ethernet.h is included:

#include <Arduino.h>
#include <Modbus.h>
#include <SPI.h>
#include <Ethernet.h>

I received some Errors by compiling. Changing to 
#include <Ethernet2.h>  
solved this issue.

wvengen, 2018-06-05:
I've gotten this to work with https://github.com/wvengen/modbus-arduino