andresarmento/modbus-arduino #35: Reading 8 Ists & Coils and writing multiple register bug fix

umarsaid, 2018-03-08:
I found some bugs in the library. You reading Ists and Coils for 8 registers, the 8th register will always be 0. Also, when we write multiple holding registers, the last value will be lost.

I have fix the bug in my local copy. I want to submit the bug but I do not know how.

masrodjie, 2018-03-15:
you can fork the repo, commit your changes and pull request. or you can attach your update here, i'll try to make pull request to make this library better

janchower, 2018-04-24:
I am also seeing this same issue - every 8th coil is not working.  
Umarsaid, is there any way at all for you to post or forward me the updated library you mention that you fixed?  My programming skills are still improving and I'm having trouble reading through the library to identify where the issue is.  Thank you very much in advance for your help!!!!

petersoltys, 2019-02-15:
Please would you change these lines 329 and 380 in [modbus.cpp](https://github.com/andresarmento/modbus-arduino/blob/master/libraries/Modbus/Modbus.cpp)
from
`	while (numregs--) {
        i = (totregs - numregs) / 8;
`
to
`	while (numregs) {
        i = (totregs - numregs) / 8;
        numregs--;
`
by my findings only this decrement order need to be changed.

janchower, 2019-02-15:
Thank you!!!!  I will try this as soon as I can, hopefully later today.  I’ve spent hours trying to study the library but it is clearly above my skill.  This will help me tremendously!

I’ll email back as soon as I confirm this is OK.  Thank you very much for your help!

Best regards,
-Richard White

This system contains confidential and copyrighted information.  Access to the system is limited to users only and only for approved business purposes.
Anyone obtaining access to and using this system acknowledges that all information on this system including but not limited to electronic mail, word processing, directories and files, constitutes private property belonging to the Company.
Anyone using of viewing this system is further advised that the use of this system may be recorded and the information contained herein may be monitored, retrieved and reviewed if, in the Company’s sole discretion there is a business reason to do so.
If improper activity or use is suspected, all available information may be used by the Company for possible disciplinary action, prosecution, civil claim or any remedy or lawful purpose.

janchower, 2019-02-15:
Peter - THANK YOU - I tried this and it looks like it works!  I am very grateful for your help.  This is definitely something very important to my projects.   I really appreciate this!

raf59, 2019-07-17:
Thank you Peter !!
I met the same issue with the writeMultipleCoils function.
I did the same change at line 499 with numoutputs and it works now.

`while (numoutputs) { i = (totoutputs - numoutputs) / 8; numoutputs--;`

HGQ21102, 2019-11-11:
> Thank you Peter !!
> I met the same issue with the writeMultipleCoils function.
> I did the same change at line 499 with numoutputs and it works now.
> 
> `while (numoutputs) { i = (totoutputs - numoutputs) / 8; numoutputs--;`

Thank you, you are the best

> Please would you change these lines 329 and 380 in [modbus.cpp](https://github.com/andresarmento/modbus-arduino/blob/master/libraries/Modbus/Modbus.cpp)
> from
> `	while (numregs--) { i = (totregs - [...]

Thank you, you are the best