andresarmento/modbus-arduino #41: Modbus Ip - It works but with Timeout Error

asdelx1, 2019-06-01:
Hello ! I could connect an arduino Due with a w5100 mini ethernet shield. I can use the entire library. I use modbus poll or radzio for a modbus tcp ip master.  The problem is that for example if I read a hold reg, i get a timeout error or ( write error read error ) and then I can read. So it is Yes-No-Yes-No-Yes-No. It´s like i lose connection, maybe something about times. I use mudbus library and mgsmodbus and I didn´t have that problem !

[image]

As you see , i can run it OK but first an error and then it goes, error.. goes.. error .. goes ? Don´t you think it can be a time problem at the library ?

mail : matu_1234_ches@hotmail.com

mlednicky, 2021-02-23:
You need to uncomment line "#define TCP_KEEP_ALIVE" in used library.
If you read "README" file, you would knew.

SuriyaStar123, 2021-12-20:
nope