andresarmento/modbus-arduino #56: W5500

HVisMyLife, 2021-03-22:
Can I use W5500 module the same way as W5100 shield?

rajsalieabs, 2022-05-16:
@HVisMyLife any luck? Please let me know if you were able to achieve this.

percramer, 2022-05-16:
I am using w5500 modules without any problems

rajsalieabs, 2022-05-16:
@percramer are you using W5500 for Modbus TCP? What is the host controller? Arduino (Atmega or ESP32?) Can you please share code / library references please. I badly need help.

rajsalieabs, 2022-05-17:
> I am using w5500 modules without any problems

Hi @percramer can you share the code samples / references and the library used? I badly need help.

percramer, 2022-05-17:
ESP32 with arduino. The modules I am using are the https://shop.m5stack.com/products/lan-module?variant=16804778803290

don’t have code samples right now because I am at work. I can send those later

rajsalieabs, 2022-10-11:
> From "Per Cramer" ***@***.***>
> To "andresarmento/modbus-arduino" ***@***.***>
> Cc "RAJ VIGNESH Navaneetha Krishnan" ***@***.***>;
> "Comment" ***@***.***>
> Date 17-05-2022 02:17:23
> Subject Re: [andresarmento/modbus-arduino] W5500 (#56)

Would you mind sharing the code and library? I badly need guidance. 
Thanks.

​——

Regards,

RAJ VIGNESH T N,

SALIEABS Electronics Engineers LLP,

Ph:- +91 9791848808